psychogenic/SerialUI #1: software serial

ronbal, 2015-03-17:
Hi can i use Software serial or the serial1 on leonardo?

psychogenic, 2015-03-18:
Greetings Ron,

On 15-03-17 04:36 PM, ronbal wrote:
>
> Hi can i use Software serial or the serial1 on leonardo?
>

The SerialUI constructor accepts a 3rd parameter--a pointer to the 
Stream* object--to set which serial port to use.  You can check the 
SerialUI.h file for details  but, in short, you should be able to replace:

    SUI::SerialUI MySUI(device_greeting);

with

   SUI::SerialUI MySUI(device_greeting, 0, &Serial1);

If you want to use some kind of software serial, you may need to do some 
casting (depends on whether it's derived from Stream or not).

If you have to fiddle with things to get it working with the software 
serial, I'd appreciate if you could share your results (which I'll 
eventually update the docs with).

Cheers,
PatD

ronbal, 2015-05-17:
I Need your help again.

i have tried to compile mit Code for the Arduino due but where are many 
errors. i use Subversion 13.

C:\Users\Ronny\Documents\Arduino\libraries\SerialUI\SerialUI.cpp: In 
member function 'size_t SUI::SerialUI::print_P(const char*)':
C:\Users\Ronny\Documents\Arduino\libraries\SerialUI\SerialUI.cpp:459:69: 
error: 'strncpy_P' was not declared in this scope
   strncpy_P(p_buffer, message, SUI_SERIALUI_PROGMEM_STRING_ABS_MAXLEN);
^
C:\Users\Ronny\Documents\Arduino\libraries\SerialUI\SerialUI.cpp: In 
member function 'size_t SUI::SerialUI::println_P(const char*)':
C:\Users\Ronny\Documents\Arduino\libraries\SerialUI\SerialUI.cpp:468:69: 
error: 'strncpy_P' was not declared in this scope
   strncpy_P(p_buffer, message, SUI_SERIALUI_PROGMEM_STRING_ABS_MAXLEN);
^
C:\Users\Ronny\Documents\Arduino\libraries\SerialUI\SerialUI.cpp: In 
member function 'void SUI::SerialUI::showTrackedState()':
C:\Users\Ronny\Documents\Arduino\libraries\SerialUI\SerialUI.cpp:576:59: 
error: 'dtostrf' was not declared in this scope
      totlen += strlen(dtostrf(dval, 5, 2, &(outBuf[totlen])));
                                                            ^
Fehler beim Kompilieren.

psychogenic, 2016-01-08:
Wow, took me a while to see this (sorry).  
In short, there was a "Due" flag in the SUIConfig.h -- until today: now you edit the SUIConfig and comment out the default platform define, setting instead "#define SUI_PLATFORM_ARDUINO_DUE", here:

/*
	 * SUI_PLATFORM_X
	 *
	 * Define ONE OF the available SUI_PLATFORM_X values to select the serial
	 * implementation platform.
	 *
	 * SUI_PLATFORM_ARDUINOSTANDARD: regular Arduino/Arduino-compatible.
	 *  Is the default and will usually just work(tm)
	 *
	 * SUI_PLATFORM_ARDUINO_DUE:	What it says, Arduino Due.
	 *
	 *
	 * SUI_PLATFORM_RBLNRF51822: nRF51822-centered Arduino SDK, by
	 * RedBearLab, for things like the BLE Nano http://redbearlab.com/blenano/
	 *
	 *
	 * SUI_PLATFORM_DIGISPARKUSB:  DigiSpark, needs testing/work.
	 *
	 * NOTE: for SUI_PLATFORM_DIGISPARKUSB, I've had to add a
	 * 	#include <DigiUSB.h>
	 * at the top of my sketch to get it to compile at all...
	 */
	#define SUI_PLATFORM_ARDUINOSTANDARD
	// #define SUI_PLATFORM_ARDUINO_DUE
	// #define SUI_PLATFORM_RBLNRF51822
	// #define SUI_PLATFORM_DIGISPARKUSB

closed 2016-01-08 by psychogenic