psychogenic/SerialUI #4: Deprecated SovA usage on 1.8.2

VitorBoss, 2017-06-07:
The code don't compile on Arduino IDE 1.8.2
`sketch\serialmenu.h:56:13: warning: This is deprecated SovA usage...

 SUI_DeclareString(greeting,"+++ Welcome to the ArduStim +++\r\nEnter ? for help");
             ^
sketch\serialmenu.h:57:13: warning: This is deprecated SovA usage...

 SUI_DeclareString(bytes, " bytes.");`

psychogenic, 2017-06-07:
Well, the deprecation warning shouldn't kill the compilation, in theory, but it _is_ deprecated... where is this code from?

Have you checked the SerialUI examples--they should demo how the same is accomplished with the revised API.

VitorBoss, 2017-06-07:
Ardustim

psychogenic, 2017-06-07:
AFAIK, Ardustim has yet to be ported to SerialUI v2... I gave it a quick go, but ran into memory issues, so pushed it back until later.

In the meantime, remove your SerialUI install and replace with v1.4 from:
https://github.com/psychogenic/SerialUI/archive/v1.4.0.zip

Should work fine.