psychogenic/SerialUI #10: SerialUI V3 does not compile with Arduino Leonardo/Micro/Pro - ATMEGA32U4 microcontroller

Matthias-Germany, 2019-09-10:
First of all, it's an interesting project. I am curious to see it working.

While V2 compiles without errors on the Arduino Leonardo/Micro/Pro - ATMEGA32U4 microcontroller, V3 creates the following error:

`cannot convert 'Serial_*' to 'SerialUI::Comm::SourceType* {aka HardwareSerial*}' for argument '2' to 'SerialUI::SerialUI::SerialUI(uint8_t, SerialUI::Comm::SourceType*)'`

I will attach the full description below the message. 

I think it hast to do with the serial interface being native USB in this microcontroller. If I switch the 

`#define SUI_PLATFORM_SOURCE_DEFAULT		Serial`

to Serial1 (which is the second available serial and a standard serial interface on this controller), the compilation is done without an error. Unfortunately then there is (of course) no data via the USB interface. With the standard setting 'Serial' (which is the USB serial) the above error is created.

Maybe it's just a small issue and can be solved by casting the pointers, but I am not knowledgeable enough to find a workaround. But I am willing to try a few workarounds.

Attached the full error message. 

Arduino: 1.8.9 (Windows 10), Board: "Arduino/Genuino Micro"

In file included from C:\Users\212748107\Documents\Arduino\libraries\SerialUI-3.1.0\src/SerialUI.h:36:0,

                 from sketch\callbacks.cpp:42:

C:\Users\212748107\Documents\Arduino\libraries\SerialUI-3.1.0\src/includes/SerialUI.h:60:62: error: could not convert '& Serial' from 'Serial_*' to 'SerialUI::Comm::SourceType* {aka HardwareSerial*}'

    Comm::SourceType * commsrc = &(SUI_PLATFORM_SOURCE_DEFAULT)) ;

                                                              ^

In file included from C:\Users\212748107\Documents\Arduino\libraries\SerialUI-3.1.0\src/SerialUI.h:36:0,

                 from C:\Users\212748107\Documents\Arduino\libraries\SerialUI-3.1.0\examples\BasicUI\BasicUI.ino:42:

C:\Users\212748107\Documents\Arduino\libraries\SerialUI-3.1.0\src/includes/SerialUI.h:60:62: error: could not convert '& Serial' from 'Serial_*' to 'SerialUI::Comm::SourceType* {aka HardwareSerial*}'

    Comm::SourceType * commsrc = &(SUI_PLATFORM_SOURCE_DEFAULT)) ;

                                                              ^

In file included from C:\Users\212748107\Documents\Arduino\libraries\SerialUI-3.1.0\src/SerialUI.h:36:0,

                 from sketch\setup.cpp:42:

C:\Users\212748107\Documents\Arduino\libraries\SerialUI-3.1.0\src/includes/SerialUI.h:60:62: error: could not convert '& Serial' from 'Serial_*' to 'SerialUI::Comm::SourceType* {aka HardwareSerial*}'

    Comm::SourceType * commsrc = &(SUI_PLATFORM_SOURCE_DEFAULT)) ;

                                                              ^

setup.cpp:54:22: error: cannot convert 'Serial_*' to 'SerialUI::Comm::SourceType* {aka HardwareSerial*}' for argument '2' to 'SerialUI::SerialUI::SerialUI(uint8_t, SerialUI::Comm::SourceType*)'

 SUI::SerialUI MySUI(7);

                      ^

exit status 1
cannot convert 'Serial_*' to 'SerialUI::Comm::SourceType* {aka HardwareSerial*}' for argument '2' to 'SerialUI::SerialUI::SerialUI(uint8_t, SerialUI::Comm::SourceType*)'

Dieser Bericht wäre detaillierter, wenn die Option
"Ausführliche Ausgabe während der Kompilierung"
in Datei -> Voreinstellungen aktiviert wäre.

niheij, 2023-07-29:
Hi @Matthias-Germany 
I was wondering if you had solved this problem? Because I want to work with this library but the same error happens.
I would be very happy if you found a solution?
Or maybe the author @psychogenic can help with a solution on this problem?

Thanks in advance :)

Matthias-Germany, 2023-07-30:
Unfortunately I cannot help any further than my comment above. I did not resolve the problem itself. I'm not quite sure how I handled it in 2019, but probably I switched to another AVR with a separate serial-to-USB controller. Another possibility is using a menu system for AVR that also supports serial output, i.e. https://github.com/neu-rah/ArduinoMenu.